vehystrix/Elegoo-Neptune-marlin #1: Neptune 3 Pro reboots and resets EEPROM after leveling bed with LCD.

L0st-Moth-of-Indigo-Swamp, 2025-04-14:
Every time I level the bed from the LCD with this firmware after I set the Z offset and exit my printer reboots and all my settings are deleted. I reverted to the official software (V 1.4.2 - 1.x.5.1b) and found the same issue. I also reverted the screen software to the official release (1.4.2). It wasn't until I switched to the V1.x.5.1c and screen firmware 1.4.3 branch that the issue seems to have disappeared. I'm not sure what the issue was but it may be that the official software (screen or board) was broken when Hummtaro originally forked it.

awilliamwright, 2025-05-29:
> For what it's worth, I rolled back to the last version Hummtaro released, based on 2.1.2.3 and it seems to have corrected the issue. I'm going through the code to see if I can ID the root of it, but I'm still new to this and it may take a bit.

I was having this happen to me before I'd done anything to the firmware, just using the last update from Elegoo on my Neptune 3 Plus. It wouldn't happen every time, so I don't know what the x-factor was that caused it. But that was only one of many awful problems I was having with that firmware. 

Now that I've installed "2.1.2.3 build 1," which I believe is the version Waffqle is referring to, I'm now having it happen every single time I try to auto level the bed. 

HOWEVER! I discovered that if I let it auto-level the bed, then right after it finishes, I TURN OFF the printer, it will save the bed leveling information. Then I can go in and tune the z-offset, and it won't crash.

vehystrix, 2025-05-30:
Here's the release:
https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250516