vehystrix/Elegoo-Neptune-marlin #3: Known cases that trigger EEPROM reset / wipe / default settings

SweeperB0t, 2025-04-14:
This list contains all the issues already posted here plus several others of my own finding.

Printer: Neptune 3 Pro
Screen Firmware: "ELEGOO - V1.4.2-20230520" (https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/blob/main/screen%20firmware/V1.4.2-20230520/3D38_20230520.tft)
Board Firmware: "vehystrix - bugfix-2.1.x (2024/12/05" (https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20241205)

1. Attempting to set Z-Offset in the "Level" menu WITHOUT initiating the "Auto Bed Leveling" afterwards triggers a reset.
       a. performing ABL, followed by messing with Z-Offset will still trigger the issue. ABL MUST be performed after adjusting Z-Offset.

2. Opening the "Settings" menu and attempting to enable the "LED" triggers the reset.

3. Tapping the "Print" button with no SD card inserted, followed by acknowledging the error message about no SD card being inserted, will trigger the reset.

This is not an exhaustive list as I'm sure there are several other ways to trigger reset. Share below anything else weird noticed or other triggers as to help possibly pinpoint the issue. Anything helps. I'm not savvy when it comes to firmware so I'm just starting this thread to collect all issues in one spot

kclambi1, 2025-05-16:
Can confirm this still occurs, even after restoring default M502 and saving M500.  As soon as I adjust the z height from the touchscreen, even without doing any actual leveling, once I return back to the home screen the firmware crashes and restarts.  Reproduced every time, multiple SD cards, starting from scratch default settings.  This is of course as you stated using the touch screen as the primary controller rather than octoprint or a peripheral.

vehystrix, 2025-05-16:
I just tried this and indeed it crashes the machine, but not every time (I tried it 4 times and it crashed the first time and didn't the next 3 times).
I have no idea why this is happening.
Errors are being triggered here https://github.com/vehystrix/Elegoo-Neptune-marlin/blob/a8bec871065b86ea6dd08e29e7b29263c4389bf2/Marlin/src/HAL/STM32/eeprom/eeprom_wired.cpp#L55

I know that it's not related to the LCD, it's just that exiting the level menu triggers a full settings write. I've had it crash when running `M500` via USB as well.

For me the issue is obviously no showstopper because it only happens when saving settings which is something I rarely have to do. And if it does happen to crash I just restore all settings from my settings GCODE file. (I did on occasion have to run it twice because the first time it triggered the crash)

I've checked the code and there's no difference between the 1.x.5.1c code from Elegoo and the code I have now (at least for the writing to EEPROM code)
If anyone could help me figure out how to debug this thing, because I'm kind of stuck. For now I just added output on every EEPROM write, to see if I can find a pattern in what writes are failing.
So if you want to help, I'll upload a build with this enabled.

kclambi1, 2025-05-16:
Sure!  Feel free to upload the build.  I can check that against some other files I have that do no exhibit that crash and see if I can help narrow it down.

vehystrix, 2025-05-17:
Builds with debug output for the EEPROM enabled:
https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250516

LeoZanellaM, 2025-06-07:
Hello,
I hope this helps.
I got the error first and tried to reproduce it, attached is the dump.
What i did is, right after installing the firmware, I adjusted the e steps then tried to level my bed (i know, my numbers are terrible, I am waiting for an adjustment wheel set).
if you guys need i can reinstall the firmware and try to get more data, 

[Neptune3ProError.txt](https://github.com/user-attachments/files/20640472/Neptune3ProError.txt)

vehystrix, 2025-06-08:
Looks like it's failing on the first value it's actually trying to write (your adjusted e-steps)
No idea why, but I can think of 2 reasons:
1) These machines have flaky EEPROM that just bugs out sometimes
2) The arduino STM32 code to write to wire EEPROM has changed between 2023/05/18 (v1.x.5.1b) and now, causing writes to (sometimes) fail on our machines.

If it's case 2, compiling the code from 1.x.5.1b now would also cause EEPROM write crashes. This is something we can verify.
If it's case 1, we're not going to be able to fix it...

shaydin, 2025-06-13:
Just dropping in to say I'm having the same issue. For me, it seems to be happening on every print when printing from OctoPrint. I also rarely use the screen and don't have issues with it when I do perform bed leveling and Z-Offset calibration. But when I print, it fails at the end of each print and resets the settings. 

I don't know that it's #1 - as much as I hate to defend Elegoo after some issues with their customer service on another printer. This only happened after I flashed this updated firmware. I'll flash back to stock and see if it continues and report back if you'd like.

vehystrix, 2025-06-13:
When you say it fails at the end of every print, at what command does it fail?
Can you share the gcode of a (simple) print that I can use to try and reproduce the issue?

shaydin, 2025-06-16:
> When you say it fails at the end of every print, at what command does it fail? Can you share the gcode of a (simple) print that I can use to try and reproduce the issue?

I'm not sure. Seemed to be when the print fully completed and the extruder was moved back to its safe end coords. I haven't been able to print a single thing since adding in my two cents because I seemed to have jinxed myself. I now can't get it to perform a z-offset calibration or bed leveling without the EEPROM getting wiped.

If you want worse news, I think it might be flaky EEPROM. I flashed back to stock in a fit of frustration and it happened again. Then I went and checked the dormant NARUTOfzr repository and there are several people complaining of the same problem (obviously with no reply since this printer has well and truly been abandoned).

vehystrix, 2025-06-16:
Yeah it seems this is the same issue:
https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/issues/80
So probably flaky EEPROM, nothing we can do about it then...

As for the machine crashing after a finished print, all I can think of is that there's a GCODE at the end that triggers an EEPROM write. Does any of you have logs of the serial console from when this happens, that will give a good indication what is triggering the crash.

kclambi1, 2025-06-16:
I had a log, but lost it when pasted into Sublime for upload then MacOS decided to update.....But, I did find that when using Prontoface, the v1.4.3 NarutoTZ screen firmware and your latest release, most all EEPROM writes were fine, including leveling.  It was when I attempted to go through my usual tuning regiment (linear advance, PID nozzle, bed modeling etc) that I noticed if I tried to write a different k-value for linear advance, it crashed and erased all settings.  This did not occur with any other PID or calibration configuration values, and the log did not suggest anything was buggered until that point.  I was able to replicate this thrice in a row.

shaydin, 2025-06-16:
Possibly a dumb question to the entire audience here, but I have a spare BTT motherboard for an Ender 3 Pro I'm no longer using. SKR E3 V3 I believe. I imagine it's not as simple as replacing the faulty Neptune mobo with the SKR and flashing the Netpune Marlin firmware, correct? The naive hope is that as long as its got the correct specifications from Marlin it would work (not including the touch pad), but something is telling me there's more to it than that. 

Just trying to find a solution around the bad EEPROM that doesn't involve buying another Neptune board that will have the same problem later on. 

If the lift for the SKR involves engineering new firmware entirely, then does anyone know of a more reliable board replacement than the OEM one?

vehystrix, 2025-06-16:
You should double-check that the mobo you have has the correct output pins, other than that, you'll have to flash either the ender firmware for the board, or compile the source here for the model of your board (that way the LCD will probably work if you can connect it to the board.

I think the easier way if you're willing to forgo the LCD is to just flash klipper and run that. No need to write to EEPROM with clipper

vehystrix, 2025-06-17:
Crashing when turning on the hotend sounds really weird.
Are you sure it's the same crash? Any logs from the serial console?
Sounds like you got a flaky power supply or motherboard.
The only pattern I can see for now is that these machines have great hardware, but flaky electronics. so when they work, they work really well, and when they don't it's a mystery as to why it fails.

I'll say it again. without logs from the serial console any crashes you guys are reporting give us no real info.
The serial console will have *some* info as to why the printer decided to restart itself, from errors writing to EEPROM to faillures to hit endstops to heating issues (eg. temperature dropping while the heater power is on)
I've had all of these crashes before, and every time there was in the serial console the exact reason as to why it crashed.

shaydin, 2025-06-18:
I can try to grab some logs from a crash. I haven't updated to Klipper yet and need to get evidence of the failure for Elegoo support. Amusingly it came up while getting a warranty replacement motherboard for my new Mars 5 Ultra, and they also want to take a look despite it being out of warranty. You're dead on that they have excellent hardware, but very poor electronics.

vehystrix, 2025-06-18:
The only way to connect logs is to attach a PC via USB and save the output of the serial console there.
Even better is if you also print via the serial console, that way the gcode that gets sent to the printer is also in the log and you get an indication of what command crashed the printer.

The easy way: Install [Octoprint](https://octoprint.org). You can install it on your windows PC, a raspberry pi, an old android phone, etc...
Then upload the gcode file from your slicer to it and print from there. Also turn on serial console logging (remember to clean up logfiles regularly, they get very big very fast!)

vehystrix, 2025-06-23:
Few questions/remarks on the log file:
You ran `G28` `M500` `G28` `G29` `M500` `G28` `M500` `G28` `G28` and then pulled out the SD card, correct?
That's a lot of homing, bed leveling and saving to EEPROM. I'm going to assume this is to try and trigger the crash?
You compiled from source yourself? The only changes are enabling the N3MAX define and changes to the steps/mm for X&Y and input shaping frequencies? You did not enable `PRINTCOUNTER`?

I can't really see a clear reason why it's crashing, *especially* because it has no reason to be writing to EEPROM at the end anyway...
2 things you might try:
Enable logging to SD with [`M928`](https://marlinfw.org/docs/gcode/M928.html)
Build with the extra defines `DEBUG_EEPROM_READWRITE=1` `DEBUG_EEPROM_READWRITE_EXTRA=1`

With this we can hopefully notice any `M500` commands that get inserted into the queue automatically (eg not because they were sent via serial)
And with the debug defines every EEPROM write attempt (in the `settings.save()` function) will be logged to the serial console, so we know what is being written even if it's not triggered by an `M500` command.

If there's no extra `M500` commands and no logging of EEPROM writes but it still crashes, there's another place in the code somewhere writing to EEPROM that I haven't found yet...

calebsexton, 2025-06-26:
I changed the size of the bed mesh from 7x9 to 7x7. The error I kept seeing in Octoprint was "Failed to enable bed mesh" or something like that. 

The only thing I know about C++ memory management  is to stay away from it., but my hunch is that is where the problem is. My hope was by shrinking the mesh I could avoid writing to or reading from the memory causing the reset.

I haven't had any EEPROM resets since on my N3 Max. I have a working bed mesh and have sucessfully printed a part.  It's too early to tell if that was the culprit, but I hope it was. 

I'm still working on getting my logging build to write to the SD. I keep getting an error on the touch screen about writing to card. I'll keep at it.

shaydin, 2025-06-27:
I would encourage everyone to open a ticket with Elegoo by emailing 3dp@elegoo.com even though the printers may be out of warranty. This is most likely a design flaw, and they need to be aware that it's not as simple as replacing the motherboard if that's the case.

In my ticket, they admitted that the motherboard has to be the problem (after some of the worst customer service I've ever had to deal with), but pushed me to buy a new one. Since this is more than just a one off, I have no confidence that the replacement won't have the same flaw.

If you reference my ticket, which is already 3 escalations up, it may help. Ticket number is 349618. Making them troubleshoot all of these cases is really the only way to get any kind of traction with them, it seems.

rolandjutai, 2025-08-14:
> I changed the size of the bed mesh from 7x9 to 7x7. The error I kept seeing in Octoprint was "Failed to enable bed mesh" or something like that.
>
> The only thing I know about C++ memory management is [...]

Is this still working for you? What I notice on my N3 Pro is that the mesh is generated by probing the very edge of the right side, but it doesn't go all the way to the left. Is this correct? I do wonder if there's something going on with the bed mesh sizes / code? Maybe I'll try lowering mine to 5x5, see what happens...

vehystrix, 2025-08-14:
The printer uses a probe that's located 22mm back and 28.5mm to the left of the nozzle. This is why when probing the nozzle isn't on the edge of the bed on the left side, but is on the right side.

For my N3 Plus the probe points are:
X: 15 | 62.75 | 110.5 | 158.25 | 206 | 253.75 | 301.5
Y: 22 | 70.83 | 119.67 | 168.5 | 217.33 | 266.17 | 315
See that on the left side it probes at 15mm from the edge, but on the right side it probes 28.5mm from the edge. On the front of the bed it probes 22mm from the edge, and on the back of the bed it probes 15mm from the edge. If you make a mesh of 5x5 you'll still see these same offsets on the left, right, front and back of the bed. The distribution between these will just change

vehystrix, 2025-08-14:
For us the EEPROM reset doesn't work because for them it also doesn't work...
The issue they describe is when upgrading from the b to the c revision of the official firmware, suddenly settings get different values because the EEPROM layout changed. For this, resetting EEPROM does indeed work.
In the firmware I build the reset always happens when you flash it (to avoid this exact issue)

The thing is that the original firmware has a lot less settings to write to EEPROM, so less writes means less chance for the EEPROM to suddenly crap out. I'm still convinced it's faulty hardware at the core here, maybe even temperature related, and not some software issue.

rolandjutai, 2025-08-14:
Got you! Well, it does happen on my N3Pro even when cold. But I understand what you’re saying. It could be that we hit some sort of bug by writing to areas that don’t quite work. 

That’s a bugger. But such is unofficial firmware. Appreciate all your hard work. :-)

rolandjutai, 2025-08-15:
Reverted to [bugfix-2.1.x (2025/04/12)](https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250412) and it's crashing less for me. It was left right centre with latest build on my N3 Pro. With this one, less crashing so far. Time will tell. I'll report back.

rolandjutai, 2025-08-17:
Ok no worries. Let’s see if I can find a build first where the crashes aren’t occurring. That might help us track it down. 🧐

rolandjutai, 2025-08-25:
I did a new build with MeatPack enabled, but I haven't been able to test it yet as I'm back home around Thurs / Fri this week. It's here, but use at your own risk, guys... not yet tested at all. I'll install and test when I get home. https://github.com/rolandjutai/Marlin-Neptune-3-Pro/releases

vehystrix, 2025-09-09:
So the flaky hardware theory is still the most likely to be correct...

vehystrix, 2025-09-09:
> Right... a crash at the end of the day, but it was stable longer for sure. Just a thought...
> 
> Could we do something like:
> 
>     1. In Configuration_adv.h
>        #define ENABLE_LEVELING_AFTER_G28
>        #define RESTORE_LEVELING_AFTER_G28
> 
>     2. In Configuration.h:
>        #define EEPROM_INIT_NOW
> 
> 
> Then maybe we can just run M420 S1 at startup and load a mesh without having to run G29?
> 
> OR
> 
> in leveling.cpp
> 
> planner.leveling_active = true;
> 
> One of these should work to be able to load a mesh from either an EEPROM restore or simply by using G29 W commands without a full G29 run. That's the most annoying part of these crashes.
> 
> Thoughts?

Added a solution for this in https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250909

rolandjutai, 2025-09-09:
> Added a solution for this in https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250909

Wow, that's amazing man!!! That's really awesome. I might move back to your firmware just for this as it does save a lot of time after a crash. I could even query whether mesh is enabled before a print, and if not, restore the mesh, then print. :-) 

Do you have MeatPack enabled on your version? I forget. If you could enable it for the next release, I'll be very happy. But not urgent, just if you get to do another release at some point.

rolandjutai, 2025-09-09:
> So the flaky hardware theory is still the most likely to be correct...

Well, it's difficult to know really. It is possible that pushing the EEPROM to its limits indeed unearthed some sort of issue with the HW. It def happens less, the less it's pushed (older firmware).

rolandjutai, 2025-10-19:
> > Do you have MeatPack enabled on your version? I forget. If you could enable it for the next release, I'll be very happy. But not urgent, just if you get to do another release at some point.
> 
> I just made a new release with meatpack enabled

Amazing! Thank you so much. Moving over to your release. :-)

valkuc, 2026-01-20:
Finished successfully, without any errors. Unfortunately, can not attach any logs as I deleted previous log files and looks like Octoprint decided not to recreate them. Guess I need to reboot my Orange Pi. Will do more tries tomorrow.

vehystrix, 2026-03-06:
None of the commands `M108`, `M77` or `M84 X Y E` should cause firmware writes.
The only thing I can think of is that the LCD code somehow triggers a settings save when the print timer stops (and thus the LCD transitions from the progress screen to the main screen)
I'll add more debug output to check for this when I get around to making a new release.

vehystrix, 2026-03-29:
New builds here:
https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20260329

I added output to serial whenever `settings.save()` is called from within the UI code, so we can see where the EEPROM writes are getting triggered after a print (see https://github.com/vehystrix/Elegoo-Neptune-marlin/commit/3d219809e23e5c2b703cae48c53e53dc19efb158)

vehystrix, 2026-05-16:
Hey, just make a PR if you manage to fix the EEPROM issue, and I'll add it to a next release.
Put the wifi implementation in a separate PR if you can, this will help keep thins clean.

Thanks for putting in the effort and helping out!

@valkuc  I'll see if I can find some time next weekend to publish another release

vehystrix, 2026-05-20:
I went trough the complete code to see where there's EEPROM writes (they all go trough the same HAL calls so they're easy-ish to trace back) and the only place we're writing to EEPROM is in the `settings.save()` function. This function is called from `M500` (obviously) and also directly from a bunch of LCD actions (each of which now also have trace logging in the debug builds)

So why the EEPROM would corrupt itself at random times is unknown but reeks of flaky hardware...