vehystrix/Elegoo-Neptune-marlin #5: [BUG] (bug summary)Bugfix 2.1 both March and April versions

MadAriduckwo, 2025-04-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying the Bugfix 2.1.x versions for March and April, when doing an automatic bed level, once the leveling completes and the extruder moves back to the home position it does not stay in the z position to allow for manual z offset adjustment. It rises a few millimeters higher negating the option to adjust properly. If I plus in my previous z axis reading using Marlin 2.1.2.5b it prints satisfactorily even though the head is too high based on its position after the ABL.

Additionally, on the April bugfix 2.1.x upon completion of a print rather than retracting the filament it appears that it unloads it which causes the next print to start with an unprimed nozzle.

### Bug Timeline

new bug in april and carryover bug from march bugfix 2.1.x

### Expected behavior

Expect nozzle to remain in z offset position following ABL confirm to allow for manual adjustment.

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Elegoo Neptune 3 Pro

### Electronics

Stock

### LCD/Controller

V1.43 firmware on stock LCD screen

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Using your standard configuration.h and configuration_adv.h

[configuration.zip](https://github.com/user-attachments/files/19673436/configuration.zip)

vehystrix, 2025-04-14:
It will show progress when printing over USB, but only if your slicer has included `M73` commands to indicate progress. So check your slicer settings for this.

Eg. For Orcaslicer, it's under Printer settings > Basic information > Disable set remaining print time (make sure this is unchecked to include the `M73` commands)

MadAriduckwo, 2025-04-14:
Ok, I enabled in Cura. Thanks again!