vehystrix/Elegoo-Neptune-marlin #6: Doesn't read sd card

bckyrdbshr, 2025-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After install print function no longer works. Reinstall old firmware and it works fine

### Bug Timeline

Started after install, never worked

### Expected behavior

To press the print button and it bring up files to print on the sd card

### Actual behavior

Press the print button nothing happens

### Steps to Reproduce

Installed your firmware didn't work
Reinstall old firmware works fine
Reinstall your firmware again didn't work
Installed old firmware back and is currently working fine

### Version of Marlin Firmware

2.1.2.3

### Printer model

Neptune 3 plus

### Electronics

Stock

### LCD/Controller

original that came with machine

### Bed Leveling

ABL Linear grid

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm sorry i gave you all the info i can, im not even able to hook my printer to my computer at the moment, some of the questions here i have no clue. First time ever doing this. I guess if it can't be solved with what info i've gave i guess i'll stick to my old firmware. It would be nice to try something new..

Firmware ver. 2.1.2.3b
UI ver. 1.5beta02

bckyrdbshr, 2025-05-09:
OK, I get it. Since I'm not very computer savvy my problems are just that I guess, my problems. Its very apparent no one on here gives a rats ass.

D1SK0V3RY, 2025-05-09:
I had the same problem, formatting didn't solve the problem - neither of the 2 cards worked, although they work just fine on the stock firmware. I printed via USB to test this fw, but when my settings crashed on their own it was the last straw. Returned to stock for now.

vehystrix, 2025-05-15:
Ok so first off, I'm doing this in my spare time and apparently forgot to turn on email notifications for new issues. I only just saw that you posted this. You're not on your own here and I do care, but I can only do so much.

You state you used firmware 2.1.2.3b with UI version 1.5beta02. Version 2.1.2.3b is not one I made. I got v2.1.2.4b and 2.1.2.5b, also a lot of bugfix-2.1.x releases. I stopped tracking the 2.1.2.x releases in the marlin repo a while ago.
Now whatever version you got from this repo it should work, the reason why it probably doesn't work is that UI version 1.5beta02 is not supported with this build. You need to use version 1.4.3 found [here](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/releases/tag/firmware_1.x.5.1c-UI_V1.4.3-20230707)

Check with that UI version if that fixes it. I myself usually print over USB, but the last time I printed from the SD card was the 25th of April, and it printed just fine.

D1SK0V3RY, 2025-05-15:
> You need to use version 1.4.3

That's exactly my case, I didn't update the firmware to 1.5 and stayed on the official 1.4.3, but still have this issue. I'm not sure if Naruto changed anything in your link in relation to the official version, I'll try again in a few days with your link. For now I've returned to the version from Hummtaro, it works fine. If you need more information - write to me, I'm a newbie, but I'll be glad to help, I liked your version quite more except for some points

vehystrix, 2025-05-15:
I'll check tomorrow if I can run a print from the SD with the latest build I'm running now. If that works for me I don't really know what I can do.
I assume the issue is that the print won't even start? or is it with a specific GCODE that you have issues? (I can run any file I have laying around to test?)

D1SK0V3RY, 2025-05-15:
It doesn't even open the file selection, no matter what files there, but it works fine with a USB connection, so i assume it's not gcode issue

D1SK0V3RY, 2025-05-15:
[NEPTUNE 3 PRO - APP_V1.1.5.1c-UI_1.4.3-20250425-EN.zip](https://github.com/user-attachments/files/20234651/NEPTUNE.3.PRO.-.APP_V1.1.5.1c-UI_1.4.3-20250425-EN.zip)
That's exact version of screen UI i'm used,I forgor where i found it, sry, it was weeks ago + your [bugfix-2.1.x (2025/04/12)](https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250412) [PRO_ZNP_ROBIN_NANO.bin](https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/download/N3P-bugfix-2.1.x-20250412/PRO_ZNP_ROBIN_NANO.bin)

vehystrix, 2025-05-15:
I'll check tomorrow that screen version I'm running but I assume it's v1.4.3. (I have a print running for another 16 hours, and I'm not going to cancel it over this)
I'll also verify that I can print from the SD then.

Just to be sure, you're using an SD card formatted to FAT32? Preferably the 8gb card that came with the machine (Just to make sure it's actually compatible, although I doubt it matters much)

D1SK0V3RY, 2025-05-16:
> Just to be sure, you're using an SD card formatted to FAT32?

Yeah, ofc, I tried both the one that came with the car and mine, formatted them to fat32 a couple of times when I encountered this problem

vehystrix, 2025-05-16:
Ok so I have good news and bad news:

The good news is that I just managed to print from the SD card.

the bad news is that it seems I can't reproduce your issue...

Here's what I did:
https://github.com/user-attachments/assets/386db417-3783-4e10-bdb8-c71ab18d3784

D1SK0V3RY, 2025-05-16:
> the bad news is that it seems I can't reproduce your issue...
Maybe that's because I'm using N3Pro. Idk, I'll try again as I finish my print and let u know results

bckyrdbshr, 2025-05-16:
Hi, thank you for the reply, First off no offence to you in any way. I was just frustrated and irritated. I'd been trying to get help on reddit for a totally different issue and receiving nothing but B.S. This was on an Elegoo help board, basically all i got was thats what you get for buying Elegoo. I was eventually able to get things going. There was one version I tried of the 1.4.3 UI not sure if it were the one you built, the other guy built or something totally different that if you went into setting and turned this option on, I'm Not sure what it was called at the moment but it had something to do how the files where listed on the SD card. If that was off it would work. If it were on it would break the preview of the icon and would be just a black square. What im on now seems to work like its supposed to. Thanks for what you do I appreciate all your efforts.. Sorry for the ramblings

> And I looked, and behold a pale horse: and his name that sat on him was Death, and Hell followed with him.

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

vehystrix, 2025-11-16:
So I can't get this issue to reproduce on my machine, which makes it really hard to solve...
Crazy idea: what happens if you flash the firmware for the N3Plus? Don't actually print with it (it *should* work as long as you don't go past the bed size in your GCODE, but just don't risk it imo)

As for what you can do to help, dive into the spaghetti mess that is the Elegoo UI code and see if you can figure it out...
Most of the UI code is in `Marlin/src/lcd/extui/dgus/elegoo/DGUSDisplayDef.cpp` but there's bits and pieces scattered all around, surrounded by `#if ENABLED(RTS_AVAILABLE) ... #endif` and/or `#if ENABLED(TJC_AVAILABLE) ... #endif`

Rodrigo-Cardoso, 2025-11-16:
@vehystrix thank you for the pointers. I went thru your changes and found one that tickled my interest.

I created a PR to fix this issue. Tested and working on my N3P

Rodrigo-Cardoso, 2025-11-17:
Hmm. N3pro doesn’t have that option 

[image]

chrissv, 2025-11-23:
> New release is out, can you check if this fixes it?

I just installed the release on my Neptune 3 Pro and it fixes the problem - the SD card files are now displayed.