vehystrix/Elegoo-Neptune-marlin #8: [FR] (Pausing mid printing let's the printer cool down and re heat when continuing)

leofab39, 2025-07-01:
### Is your feature request related to a problem? Please describe.

Not really, just something I feel like it should have been implemented.

### Are you looking for hardware support?

No

### Describe the feature you want

When printing, the UI screen shows an option to pause a print mid-way and continue where it left off when clicking continue.  When you pause your print, the printer will pause the printing process, then move out of the way and then it'll wait for the user to hit continue.  During this moment of waiting, the printer will maintain the heat set at the start of the .gcode, consuming electricity and burning the filament inside of the nozzle.

The feature being requested would be a change to make it so when you pause the print, it does what it normally does, but it then stops maintaining the heat and let's itself cool down or at least it lowers the heat consumed to be way less than what it normally is.  Once the user hits continue, the printer will heat up and wait until reaching the desired temperature before continuing to print again.

I do understand that letting the print bed cool down isn't the best idea as the print will most likely unstick from the surface and a main reason not to implement this.

### Additional context

I just considered that this implementation would be normal behavior on this machine and always wondered why they did it the way they did.

vehystrix, 2025-07-02:
So I checked the power usage on my printer to keep the nozzle at 205°C, and it was 20W.
I'll still look into the issue just because burning filament is bad for your nozzle, but I doubt it's going to solve your power supply issues. You could turn off your bed heating, but that will just cause the print to pop off the bed and ruin it...

vehystrix, 2025-07-07:
Fixed in this release:
https://github.com/vehystrix/Elegoo-Neptune-marlin/releases/tag/N3P-bugfix-2.1.x-20250707

closed 2025-07-07 by vehystrix